QuickLogic-Corp/qorc-sdk #62: Cannot get banner to print cleanly on first demo

tcal-x, 2020-10-16:
This is a minor thing, but maybe I'm doing something wrong.   

No matter which serial program I use, or speed I set, I can't get the "hello world" banner to print cleanly.    Either many of the characters are dropped, or I get extra line feeds and spaces.  Also, in the other direction, the app receives some garbage characters and then prints "ERROR: no such command:".

Once I get to the "[0] >" prompt, it is fine and seems very solid.

I tried using "CLI_puts" instead of "debug_str", but I'm still seeing garbling.

timsaxe, 2020-10-16:
I have not seen this before, and it is puzzling.

I assume that you are referring to qf_helloworldsw, and that the serial port is the USB.

A quick comment on the usb port.  The USB protocol is packetized, so for a direct connection such as this the baud rate is not relevant (it is for the common USB to serial cables, because those have a UART the translates the serial protocol into the packetized usb, so that UART needs to know the baud rate).  Thus as long as you are using the usb port, we can rule out baud rate as an issue.

Also, all output eventually ends up at the same driver.  If you chase thru CLI_puts and debug_str you will find they lead to uart_tx_raw() or uart_tx(), so that is why that experiment didn’t lead to any different result. Which all means that there is no obvious reason why before and after “[0]>” should be different other than the passage of time, and I don’t understand why more time would help.  You might try adding a delay before the first print.  There is already a loop to give the USB system on the host time to recognize the device.  BTW, reminds me, what OS and host?

tcal-x, 2020-10-16:
Thanks Tim,

Sorry for omitting details -- yes it's qf_helloworldsw, communicating over USB /dev/ttyACM0.  I'm on Ubuntu 20.04, and I had tried using putty and picocom.   I should try lxterm and flterm, which both come along with LiteX.  (As an aside, the ARM tools don't have a 20.04 package yet, so I needed to manually install the 18.04 .deb and also libisl15).

I have already tried a few things -- printing out some "\n\n", then another pause loop, etc.   I'll try a couple more things like that before I move on.   It is useful to know that this is not a common problem.

timsaxe, 2020-10-16:
To be accurate, most of our experience is on Ubuntu 18.04, so perhaps it is a common problem on 20.04.

coolbreeze413, 2020-10-17:
I am also on Ubuntu 20.04 and haven't noticed such problems, I use PuTTY and GTKTerm, both behave correctly.
Maybe you can try with a different USB cable and see if that perhaps is a problem.

tcal-x, 2020-10-17:
@coolbreeze413 thanks for the info!

Your comment persuaded me to install gtkwave, and with a slight change of the C code, I finally got to see the banner.

I added this to ensure a pause that the compiler can't optimize away:

```
    static volatile int vv=0;
    for (int i = 0; i != 4000000; i++) vv *= i;   // Give it time to enumerate
```

Then, if I start gtkwave, and then reset the QF board, when it comes out of reset, I see the banner correctly.

closed 2020-10-19 by tcal-x