QuickLogic-Corp/qorc-sdk #82: Build as described in Tutorial fails

turbinenreiter, 2020-12-11:
I followed the instructions to install the toolchain, but was not able to build the example as described in the [tutorial](https://github.com/QuickLogic-Corp/qorc-sdk#lesson-1a-m4-only--qf_helloworldsw).

I installed the toolchain to `/usr/share/gcc-arm-none-eabi-9-2020-q2-update/` as suggested.

I also tried using `export TC_PATH="/usr/share/gcc-arm-none-eabi-9-2020-q2-update/bin/"` to manually ensure that the right path is used.

But still the build fails with this rather cryptic message:

```
$ make
"shell Bash"
Building using GCC Toolchain
Using Linux GNU GCC Toolchain
PROJ_NAME = qf_helloworldsw
make -f makefiles/Makefile_Startup
make -f makefiles/Makefile_HAL
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/home/plam/Projects/quicklogic/qorc-sdk/qf_apps/qf_helloworldsw/GCC_Project'
Compile: /home/plam/Projects/quicklogic/qorc-sdk/qf_apps/qf_helloworldsw/GCC_Project/../../../HAL/startup/startup_EOSS3B_GCC.c
/bin/sh: /usr/bin /usr/share/gcc-arm-none-eabi-9-2020-q2-update/bin /usr/lib64/ccache /usr/share/man/man1.1.gz/arm-none-eabi-gcc: No such file or directory
make[1]: *** [makefiles/Makefile_Startup:16: /home/plam/Projects/quicklogic/qorc-sdk/qf_apps/qf_helloworldsw/GCC_Project/output/startup_EOSS3B_GCC.o] Error 127
make[1]: Leaving directory '/home/plam/Projects/quicklogic/qorc-sdk/qf_apps/qf_helloworldsw/GCC_Project'
make: *** [Makefile:40: Startup] Error 2
```

The file `/usr/share/man/man1.1.gz/arm-none-eabi-gcc` does indeed not exist, but the file `/usr/share/man/man1/arm-none-eabi-gcc.1.gz` does.

coolbreeze413, 2020-12-11:
Did you also add the toolchain into the path as below:
`export PATH=/usr/share/gcc-arm-none-eabi-9-2020-q2-update/bin/:$PATH`

If already done, could you check and post the output of 
`which arm-none-eabi-gcc` so we can figure out the issue?

jublin, 2021-01-18:
@coolbreeze413 pretty sure this is the same issue here: https://github.com/QuickLogic-Corp/qorc-sdk/issues/42

the FIND_TOOL_DIR command returns a list of available gcc-arm-none-eabi executables, and that list is sent directly as a command. So gcc-1 sees every other gcc in the list as a parameter,

tcal-x, 2021-01-23:
I just ran into this as well.

The error is quite confusing:
```
/bin/sh: 1: /usr/bin /home/tim/toolchains/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: not found
```
As @jublin also diagnosed, this is because `TC_PATH` can be set to a list.  In my case, it is `/usr/bin /home/tim/toolchains/gcc-arm-none-eabi-9-2020-q2-update/bin`.

This should be easy to fix by either just using just the first element in `TC_PATH`, or failing with a clear error if the number of words in `TC_PATH` is greater than one.

cristianpb, 2021-01-27:
In arch inux, one can find `arm-none-eabi-gcc` compiler in community libraries. They will be available in `/usr/bin` directory:
```
$ which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc
```
Then, you can compile this tutorial specifying the `TC_PATH` variable:
```
$ make TC_PATH=/usr/bin
```

jublin, 2021-02-02:
Fixed in #95, which has been merged

closed 2021-02-02 by jublin