QuickLogic-Corp/qorc-sdk #86: Example for CrowdSupply BGA64

goran-mahovlic, 2020-12-26:
Hi, is there any blink LED example for BGA64 version.
I have tried to get blink led on my custom board, but complete tool chain is focused on QFN version, so I do not know what I am doing wrong.
When I build FPGA part with changed pcf file for PD64 it just crash at startup.
DO I need to change pincfg_table as I do not see BGA version and do not know if tool chain will build good firmware if I build with current one

timsaxe, 2020-12-26:
While there are differences between QFN and BGA, since the underlying silicon is the same code that runs on one often runs on the other.  For example, we have a debug board that is based on the BGA and we run the same qorc-sdk apps on it sometimes with virtually no changes.

What is the same and what is not: all of the registers, clocking, power management, M4 to FPGA interface, is the same.  Only the pins change.  There silicon has IOs, sometimes called pads, numbered from 0 to 45.  The pin_cfg table refers to these.  All or some of the pads are bonded to balls or pins on the package.  So IO_0 might be connected to A4 on the BGA and pin#6 on the QFN (just made up, need to look at datasheet to see what is real), and IO_1 might be connected to B6 on the BGA and nothing on the QFN.

So when you have a connection to a particular ball on the BGA, you need to understand which IO that is and set the pad_cfg to configure that IO.

However, I don’t think this explains your crash.  The reason is that pad_cfg chooses what gets to control the IO: FPGA, or peripheral.  And the default is not FPGA, so if you do not configure the pad the FPGA can drive the IO but it simply won’t drive the pad.  Similarly if it tries to read the IO it will only get some internal value.  Not enough to cause a crash.

In my experience what often cause a crash is that I did start the clocks, or the FPGA is not powered up.  Under either of these conditions, the crash happens because the fist access to the FPGA fails.  Of course this assume that the M4 is trying the r/w a register in the FPGA like the advanced fpga example.

goran-mahovlic, 2020-12-27:
Thank you, then something else is wrong with my board. I do not have all those peripherals connected as qf board, for example I do not have USB, so maybe something like that is on my way. I just have LED connected to PAD_42 (H1).

I assume for blink led I would just need to enable right clock and toggle LED in main, but all samples are starting with load FPGA and starting FreeRTOS and USB serial...

timsaxe, 2020-12-28:
I am not sure what you are trying to do.  IO_42 has several potential connections:

42

0

FBIO_42

BIDIR

A0_Reg

42

1

SWV

OUT

42

2

debug_mon[7]

OUT

42

3

SDA_1_DPU

42

sensor_interrupt

IN

The table above, take from the TRM, says in mode 0 it is FPGAIO_42, and in mode 1 SWV output, and mode 2 debugmonitor7 output, and in mode 3 I2C1 SDA dynamic pullup control.

None of those let the M4 directly control that pad, so I don’t understand how you were planning to toggle the led in main.  There are 8 GPIO that can be controlled by the M4, but none are connected to IO_42.  Alternatively, there is GPIO gateware and associated app that you could configure to control FBIO_42.

goran-mahovlic, 2020-12-29:
Thank you, I will need to figure this out, and maybe wait for Qomu. Will close this issue for now, and add info if I manage to do something...

closed 2020-12-29 by goran-mahovlic