QuickLogic-Corp/qorc-sdk #91: "serial" issue in tinyfpga-programmer-gui.py

AdityaPrakash23, 2021-01-01:
Hello there!
I am try the qorc-sdk on my new QuickFeather board. I have followed all the steps in the guide but when trying the "Lesson #1a: M4 only – qf_helloworldsw", I am getting stuck at step 3(pushing the code to the board). I keep getting this error 
[image]
I tried to find the solution and tried them but nothing seems to be fixing it. I already have removed serial and installed pyserial in python3. Can someone please tell what to do? Thanks!

jublin, 2021-01-04:
Can you give the output of pip list/freeze? 

If you are using the conda environment from quicklogic-fpga-toolchain, it may not install tinyfpgab/pyserial by default (I believe it's fixed in a more recent release).

AdityaPrakash23, 2021-01-04:
I actually did a `sudo -H pip3 install pyserial` and it works now. But now it's stuck at the flash erasure stage. I am sharing the screenshot of my screen right now.
[image]
Can I redo it again? I am scared of screwing up the software of the board.

jublin, 2021-01-04:
@AdityaPrakash23 you should be able to restart it. Usually, if I get a hang, it's before the erase/program stage.

AdityaPrakash23, 2021-01-04:
@jublin thanks once again! But the qfprog command is getting stuck at the same place(the erasure command). Could u suggest what I probably could have done wrong in my setup or is there something else I could try out.

jublin, 2021-01-04:
@coolbreeze413 is the maintainer of the TinyFPGA repo. They may have some insight.

If you have a JLINK, you could try flashing all the files from https://github.com/QuickLogic-Corp/qf-initial-bins as if you were recovering a bricked device (or run bootloader from jlink and try m4 app only again)

AdityaPrakash23, 2021-01-04:
@coolbreeze413 , should I try to reprogram the board as mentioned by @jublin ? Any insight would be helpful since I am dealing with such a complex hardware after a very long gap.

coolbreeze413, 2021-01-04:
@AdityaPrakash23 The only time I have seen this kind of a problem was on a Ubuntu 18 machine which was wierdly resetting the USB connection midway - the host drivers were causing a problem, we could not track it down but a different Ubuntu installation worked ok on the same system.

Also, you should not be having to use sudo to install pyserial, the python installation may be owned/created by root perhaps?

I would suggest to try this method in a separate directory:
https://qorc-sdk.readthedocs.io/en/latest/qorc-setup/qorc-setup.html#quickstart-with-qorc-sdk

I recommend to follow the automated setup process first:
clone the qorc-sdk, and then run `source envsetup.sh` from the qorc-sdk directory as in the steps.

If you face any issues, we can walk through the manual setup steps(which is basically the same as in the envsetup script)

Please let us know which OS and version you are running on, in case you still see issues.

coolbreeze413, 2021-01-04:
> @jublin thanks once again! But the qfprog command is getting stuck at the same place(the erasure command). Could u suggest what I probably could have done wrong in my setup or is there something else I could try out. Also my green LED stays solid after pulsing for a few seconds. Could that be an indicator for something being wrong with the TinyFPGA programmer?

@AdityaPrakash23 I missed reading this carefully - the green led stops pulsing after a few seconds was exactly the behavior we saw in the one Ubuntu 18 system with the host usb driver problem. If you have access to another Ubuntu system where you could try the setup, it would be great!

Once the quickfeather board is in flashing mode (green light pulsing starts with USR button press)
check that you see the USB port in your `lsusb` list  with `lsusb | grep OpenMoko` as in here:

https://qorc-sdk.readthedocs.io/en/latest/qorc-setup/qorc-setup.html#test-the-quickfeather-usb-cdc-port

Wait for a few seconds without flashing or doing anything, check if the green LED stop blinking after around 10 seconds.

Again, check if you still see the USB port in the `lsusb` list as above.
If it has disappeared, then it is exactly the same problem we saw in the problematic Ubuntu 18 system. If this is the case, unfortunately we were not able to track down exactly why the host was resetting the USB port after a few seconds. This was not a problem with the SDK or tool installation, but with the USB driver stack, which we could not dig too deep into. You would have to try on a separate system or a separate Ubuntu installation (recommend 20.04) on the system if this is the case.

AdityaPrakash23, 2021-01-05:
@coolbreeze413  thanks for ur detailed response. I am infact using Ubuntu 18.04. As per ur last response, I set the board in programmer mode and the green Led blinks for around 7-8 seconds and then goes solid green. I check the lsusb command and I do get the message `Bus 001 Device 082: ID 1d50:6140 OpenMoko, Inc.`. Since I don't have another Ubuntu installation at hand, I will probably try the whole shebang on a Windows 10 PC. Is there any hope of making the whole thing to run on my Ubuntu 18.04 or do I need to shift my operations on Windows 10 then?

coolbreeze413, 2021-01-05:
@AdityaPrakash23 to confirm, the OpenMoko device is still present in lsusb after the led blinking stops? We have seen the LED blinking stop, not go solid green.
If you have a 3.3V USB-UART cable, could you please attach this to the QuickFeather S3 UART pins, and check the output on a Serial Program like PuTTY and post the output you see, from setting it into bootloader mode, and when the LED stops blinking.
The connections are here:
https://github.com/QuickLogic-Corp/quick-feather-dev-board#advanced

Maybe you could try hooking up a USB Hub if you have one and then try it out - am not sure if it will work, but worth a try.

You could use the Windows 10 WSL2, which should be the same thing as a native Ubuntu installation. (Windows native is not supported because the FPGA toolchain does not really run there yet, though m4 code build, and flash definitely work.)

AdityaPrakash23, 2021-01-05:
@coolbreeze413 Yes, the LED blinking stops and still lsusb shows the openmako device entry. After my disasterous experiments yesterday, whenever I opened the serial port, I was only getting garbage(automaticaly). I just managed to reflash the helloworld program and thankfully it's working properly again both in Windows and Ubuntu 18.04. Thanks a lot @jublin for sharing the initial-bins link! I did the experiment u mentioned with a CP2102 module I got with the Quickfeather board. Here is the output from reset to the green led stopping it's blinking action.
[image]
I hope this helps u in figuring out the issue with Ubuntu 18.04. I now might have to somehow start tinkering with the qorc-sdk in my Windows system probably. For that I have yet to do the setup, for which I'll probably look at the Youtube video at the Quicklogic channel. Any further suggestions? I'll probably try them out tomorrow and come back to u. Thanks a lot!

coolbreeze413, 2021-01-05:
@AdityaPrakash23 How were you able to reflash the helloworld program using the initial-bins? My understanding was that you were not able to flash the board at all, was it from a different PC?

The UART output above does not show any problems as such, so doesnt seem to be any problem with the tool, or the bootloader.

AdityaPrakash23, 2021-01-06:
@coolbreeze413 Yeah I was using my dad's PC with Windows 10. My PC has Ubuntu 18.04. I managed to flash the helloworldsw.bin from Windows 10. But I would definitely love to make it work in Ubuntu 18.04 as all my work happens on that. After so long, working on Windows feels like a real pain in the butt.
Also yes, the UART seems to be fine. Do u have any other suggestions to try out so as to figure out the exact problem? I would love to help you fix this up as it would help me too.

coolbreeze413, 2021-01-12:
@AdityaPrakash23 We went through multiple approaches to try and fix this on the 2 systems here where flashing fails, with Ubuntu 18.04, but we weren't able to fix this.
 
Strangely, 18.04 on different systems (same exact model as the failing ones) does work though.

So, we finally tried to upgrade the 18.04 install to 20.04, and the flashing now works.

I think that there is a combination of environment + hardware on 18.04 which causes this, however, we could not narrow the cause down to anything concrete.

Upgrading to 20.04 should fix the issue on your system too I believe, and you wouldn't need to have a fresh install or a different system.

AdityaPrakash23, 2021-01-12:
@coolbreeze413 Thanks for the suggestion! I will try that out and hopefully everything works out! But programming the Quickfeather is something I have found to be a bit challenging. I have my own sensors and pins to work with but haven't been able to find related documentation to interface them with the Quickfeather board. Hopefully that problem also goes away soon!

coolbreeze413, 2021-01-12:
@AdityaPrakash23 Let me know how it goes for you.

For the sensors/pins usage, you could open a separate issue, and we can look at pointing toward the right place to look, or update the documentation if its missing.

AdityaPrakash23, 2021-01-12:
@coolbreeze413 Sure! I do think the documentation is not complete. But I will probably wait for Hackster's webinar. If that doesn't work out too, then I will open a separate issue in the correct repo(probably the qorc-sdk repo). Thanks for all the help!

coolbreeze413, 2021-01-12:
@AdityaPrakash23 Yes, we are working toward having complete documentation, and any inputs you have are welcome!

timsaxe, 2021-01-12:
If the issue is working out which pins to use, you might want to look at the document for the aqi project.  I believe that goes through the process of working out which pins to use for that application, which may provide the insight that you need.

kishangondaliya, 2021-01-22:
@AdityaPrakash23 were you able to solve the issue or you upgraded to Ubuntu 20? I'd like to solve it rather than upgrading as I need Ubuntu 18 for my other projects which use some old version of packages.

AdityaPrakash23, 2021-01-22:
@kishangondaliya No I wasn't able to solve the issue. I am using Windows 10 just for flashing and also the SensiML software. I think code modification and building should be fine in Ubuntu 18.

coolbreeze413, 2021-02-08:
After some debugging, it looks like that there is a combination of hardware (PC Hardware - USB Controller) and Software (Debian/Ubuntu) that causes this kind of problem.
In some systems, Ubuntu 20.04 works ok, and in others (identical brand laptop) it does not, same with other OS and Versions.

As we have not been able to track down the root cause of this yet, as a work around, I would suggest that you use the UART Bootloader instead of the default Bootloader, using an alternative PC/Laptop where you can flash.

Note that you will need to use a 3.3V USB-UART converter and connect that to the S3 UART pins on the QuickFeather to flash images *after* changing the bootloader.
Connection details are here: https://github.com/QuickLogic-Corp/quick-feather-dev-board#advanced

Steps:
1. build the ```qf_apps/qf_bootloader_uart``` project

```
cd qf_apps/qf_bootloader_uart/GCC_Project
make
```
Copy the binary from qf_apps/qf_bootloader_uart/GCC_Project/output/bin/qf_bootloader_uart.bin to the alternative PC.

2. In an alternative PC/Laptop where you are able to flash the board (Windows/Ubuntu), flash this bootloader replacing the one on board.
Ensure that the flashing works well on this alternative  PC/Laptop (to prevent problems with flashing bootloader stuck in the middle - which might then require a J-Link to recover)

```
qfprog --port /dev/ttyACM0 --bootloader /PATH/TO/qf_bootloader_uart.bin --mode m4
```

Once this is done, from next reset the UART Bootloader will execute.

The images will need to be flashed using the USB-UART converter which will appear as ```ttyUSBx``` instead of USBSERIAL port ```ttyACMx``` on Ubuntu.

For example

```
qfprog --port /dev/ttyUSB0 --m4app /PATH/TO/M4APP/BIN --mode m4
```

If access to J-Link is available, then this can be done from any PC/Laptop even where the USBSERIAL flashing does not work.
If steps for this is needed, please let us know, we will update that here.

coolbreeze413, 2021-02-21:
@kishangondaliya @AdityaPrakash23 

It looks possibly like the modem manager is interfering with the usbserial driver, we may need to add udev rules to mitigate this: 
https://github.com/QuickLogic-Corp/TinyFPGA-Programmer-Application/blob/master/71-QuickFeather.rules

The instructions to add the rules are in the top of the file.

Could you try this on the Ubuntu/Debian sytems where flashing over the serial port does not work?

AdityaPrakash23, 2021-02-28:
@coolbreeze413 I tried following the steps that u linked. After that, when I try to set the quickfeather to flashing mode, the Green LED blinks for a few seconds, and then turns solid green and sometimes even turns off compeltely. I might have not done the steps correctly. I hope this helps.

RaghavendraKulkarni6398, 2022-07-08:
@jublin @coolbreeze413 @AdityaPrakash23  i want to write python script that will take the qfprog command and flash my board with the bin file that has been generated please help me with this idea iam struck with this for 1 week right now iam not able to figure it out.