QuickLogic-Corp/qorc-sdk #93: helloworldsw fails to build

timsaxe, 2021-01-17:
cloned qorc-sdk, cd qf_apps/qf_helloworldssw/GCC_Projects
make fails with 'gateware.h: no such file or directory'

I believe this is due to the refactoring to use eos_xyz_fpga.h

A core question is why the CI didn't catch this?

timsaxe, 2021-01-17:
Ah, probably forgot to load some submodule...

closed 2021-01-17 by timsaxe

cbc02009, 2021-02-22:
@timsaxe How did you manage to solve this? I'm having the same issue and I can't seem to figure it out.

timsaxe, 2021-02-22:
trying to recreate, but I just changed machines and that means I keeping finding more stuff that needs to be installed ... but when you cloned  qorc-sdk did you do a recursive clone?  I think that's what I forgot to do.

cbc02009, 2021-02-22:
Yup, that was exactly it. Stupid mistake on my part. Thanks!