QuickLogic-Corp/qorc-sdk #96: Can't build in WSL/Ubuntu 18.04

ArifSohaib, 2021-01-31:
I tried to build the qf_helloworld project in Windows 10 using WSL with Ubuntu 18.04. 
The error seems to be similar to issue #82 and fails with this error `c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin.exe/arm-none-eabi-gcc: not found
makefiles/Makefile_Startup:15: recipe for target `

The folder `c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin` but not bin.exe. 
I also have `C:\Program Files (x86)\GNU Tools Arm Embedded\9 2019-q4-major`

jublin, 2021-02-01:
This is fixed in https://github.com/QuickLogic-Corp/qorc-sdk/pull/95

Awaiting review. You can use the branch https://github.com/QuickLogic-Corp/qorc-sdk/tree/fix-multi-compiler-path to build until it is done.

jublin, 2021-02-02:
Fixed in #95, with is merged into mainline

closed 2021-02-02 by jublin

ArifSohaib, 2021-02-02:
Just tested it with the #95 branch and it works, thanks.