QuickLogic-Corp/qorc-sdk #100: Did not find port

just4give, 2021-02-03:
Hi,
I am trying to upload the hello world program. This is my first time using QuickFeather board. I am running on Mac OS. I was able to compile the code and generate bin file but when trying to flash, I am getting below error as in the screen shot. I am connecting the board using micro-usb cable. Can anyone help me to troubleshoot ? 

[image]

gmartin102, 2021-02-03:
Hi - Most likely this is a permissions issue with the port not having sufficient permission for users to open.  The best test of this is to try the command as sudo.  If that works then it is a permissions issue.

just4give, 2021-02-04:
Tried sudo and USB Adapter as well but no luck. Any other thoughts ? Anyone got it working on Mac?

[image]

just4give, 2021-02-04:
Was able to flash the program. I had to use cu.xxxx instead of tty.xxx like 

`qfprog --port /dev/cu.usbmodem143201 --mode m4 --m4app output/bin/qf_helloworldsw.bin`

[image]

Going to install Putty on mac to test the output.

just4give, 2021-02-04:
I am all set. 
[image]

closed 2021-02-04 by just4give

timsaxe, 2021-02-04:
> @just4give<https://github.com/just4give> , what device did you use with putty, the same one used for flashing, or different?

That will be a different device since there is a different USB device ID for the programming vs serial communication

tcal-x, 2021-02-04:
Hi @timsaxe , thanks!   I hope the info gets added to the docs to make it easier for the next MacOS user.

just4give, 2021-02-04:
The other one with tty.xxx  .... for flashing I had to use cu.xxx and for putty tty.xxx with baud rate 115200...

timsaxe, 2021-02-04:
Curious how you track this down on a mac.  On windows I run an app called device manager a look at the ‘com ports’ section.  When I plug in QF a new com port shows up and I know that is the serial port, and when I reset and bring it up in program mode that com port disappears and a different ports shows up which I then know is the programming port.  This also helps when I have more than one QF and am trying to work out which port goes with which device.

Is there a similar process that you used on a mac?

Which reminds me of a further question, in order for documentation to be useful it has to be where people naturally look for it.  Where do you generally look for answers when you have a question like this?  Github doc, github issues, forum, readthedocs, …?

just4give, 2021-02-04:
I actually did a old school process :) went into /dev folder and listed the devices issuing `ls -lrt` , then plugged in QF board and issued same command again ... new port showed up at the bottom and I new that's the one I just plugged in. There must be a better way to do this but at least worked for me for the time being.