QuickLogic-Corp/qorc-sdk #103: The source code distributed on the release page not include the git submodules.

DavidGoedicke, 2021-02-21:
Its not an important point as the instructions point towards using ```git clone --recursive```. It might just cause confusion for new users that want to make sure to use the "stable" version. 

Maybe Its possible to remove the packed source code to avoid confusion. 
[Release page](https://github.com/QuickLogic-Corp/qorc-sdk/releases/tag/v1.8.0)

Just a thought.

Best 
David

whatnick, 2021-04-04:
The environment building script `envsetup.sh` takes care of the submodule clones. Perhaps this can be added as documentation fix, that said distribution as a docker container would be great.

coolbreeze413, 2022-02-16:
@DavidGoedicke  as @whatnick mentioned, please use the envsetup.sh script for setting up the qorc sdk after cloning, or extracting the release from the archive.
The readme is also updated to reflect the steps clearly now.

Docker Container is a good idea and we will add this as well.
Please re-ooen this issue if you have any more problems with the envsetup.

closed 2022-02-16 by coolbreeze413