QuickLogic-Corp/qorc-sdk #104: Lesson-2a-fpga-only fails. Symbiflow does not write the "helloworldfpga_bit.h" file

DavidGoedicke, 2021-02-21:
[The Tutorial in question](https://github.com/QuickLogic-Corp/qorc-sdk/blob/4d009c1b52b63ab08882f427a4c138915cc15fc7/README.rst#lesson-2a-fpga-only--qf_helloworldhw)

The C compiler cannot find the ```"helloworldfpga_bit.h"``` file that was supposed to be generated by ```ql_symbiflow```.  It looks like fasm encountered an error when it was trying to write the file. 

```shell
Writing bitstream ...
Traceback (most recent call last):
  File "/home/farlab/Symbiflow/conda/bin/qlfasm", line 8, in <module>
    sys.exit(main())
  File "/home/farlab/Symbiflow/conda/lib/python3.7/site-packages/quicklogic_fasm/qlfasm.py", line 290, in main
    assembler.parse_fasm_filename(args.infile)
  File "/home/farlab/Symbiflow/conda/lib/python3.7/site-packages/fasm_utils/fasm_assembler.py", line 168, in parse_fasm_filename
    for line in fasm.parse_fasm_filename(filename):
  File "/home/farlab/Symbiflow/conda/lib/python3.7/site-packages/fasm/parser/antlr.py", line 104, in parse_fasm_filename
    parse_fasm.from_file(bytes(filename, 'ascii'), 0, callback, error_callback)
TypeError: encoding without a string argument
make[2]: *** [Makefile.symbiflow:35: build/helloworldfpga.bit] Error 1
```
---
Not sure if its relevant but I did follow the steps to install an updated version of fasm found (here)[https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain] under **Note:**

Thanks for all your help. Exsited to get going with this board :-)

coolbreeze413, 2021-02-24:
@DavidGoedicke It looks like the antlr installation went through, but due to how the fasm antlr parser works, it expects a `str` as the file path, but the toolchain actually passes in a `pathlib.PosixPath` instead, hence the problem.

Would recommend **not** to install antlr, as a combination of the toolchain with antlr has some rough edges, that have to be fixed yet.

If you do want to continue to use the setup as above, the you can fix the problem by modifying the `conda/lib/python3.7/site-packages/fasm/parser/antlr.py`

From your error, the relevant file for you would be `/home/farlab/Symbiflow/conda/lib/python3.7/site-packages/fasm/parser/antlr.py`

change the line 104 from:

`parse_fasm.from_file(bytes(filename, 'ascii'), 0, callback, error_callback)`

to:

`parse_fasm.from_file(bytes(str(filename), 'ascii'), 0, callback, error_callback)`

This needs to be done once, and after this you should be able to build and the bitstream should get generated using antlr.

Hope this helps.

whatnick, 2021-04-02:
I have hit this issue as well and will make a docs PR. To capture the antlr patch in the main docs.

whatnick, 2021-04-02:
Small update .. I have applied the antlr patch as described in my conda fasm install, however no luck with bitstream generation.
I have this error : 
```
ERROR: TCL interpreter returned an error: invalid command name "synth_quicklogic"
```
I have manually installed the toolchain. The symptoms are similar to this issue, but I suspect the root cause is different.

coolbreeze413, 2022-02-16:
@whatnick @DavidGoedicke This should be fixed now, the dependencies (specifically the symbiflow fasm) had an issue where using antlr build was failing and in addition there were changes needed in quicklogic_fasm to properly use the filepath as a string, and if you do a fresh clone and envsetup, you should no longer see antlr related issues.
Please check and re-open this issue if you still see problems.

closed 2022-02-16 by coolbreeze413