QuickLogic-Corp/qorc-sdk #106: The hello_world example not printing correctly.

ujur007, 2021-02-26:
Hi,

I tried the hello_world example and changed the string to "Hello there" from "Hello_world".

Then I build the project and uploaded it to the board. Everything went okay but finally, it did not print the right string which I had changed. Any ideas why?

coolbreeze413, 2021-02-27:
@ujur007 It is a bit difficult to figure out what went wrong without details such as the build log (output of make).
A few reasons this may happen:
- The binary did not really change (for example, while using VSCode, I add a few lines, but forget to save it, and rebuild - but nothing changes in the binary, has happened to me many times)
- The binary that was flashed was not what we intended to flash perhaps.
Otherwise, if the binary did change, and it was flashed successfully (no errors from the qfprog) then it is not possible for this behavior to occur, as the flash is erased and re-written with the new binary.

If you could do:
- reset the code to "master" branch
- build the original code - check binary size/timestamp (or better yet copy it somewhere else)
- make the changes, build the code - check the binary size/timestamp
- while doing a make, save the output to a file, with something similar to `make 2>&1 | tee build.log` - so we can check if all went ok and it did pick up the changed file on the make after making the changes

We can debug further with this info, if you are still facing a problem.

ujur007, 2021-02-28:
@coolbreeze413 
Thank you for the details on how to flash the EOS-s3. with qfprog
Actually, it was a silly mistake.There are two debug strings with "hello world" in the main file.
I only had changed one. 
Now it is working

coolbreeze413, 2021-03-07:
Closing, as no issues in the sdk to fix.

closed 2021-03-07 by coolbreeze413